leonvandenbeukel/CNC-Machine-V3 #3: Fusion 360 or STEP files

bigbass1997, 2019-10-05:
Even though STL files are adequate for 3D printing, I'd really appreciate if there was an easier way to view and edit the parts. I've noticed that you use Fusion 360 for the design work, and if you're okay with it, I think it would be a great help for everyone if you could either share a Fusion 360 public link for the project (preferable option), or alternatively export STEP files for each part.

STL files are just approximations of the original design, made up of many triangles. There's no feature history or any truly curved surface. Working with them is honestly really difficult, beyond just sticking them into a slicer program.

If you're not sure how to share a Fusion 360 design, there are instructions on [the Autodesk website](https://knowledge.autodesk.com/support/fusion-360/troubleshooting/caas/sfdcarticles/sfdcarticles/How-to-share-a-Fusion-360-design.html).

I'm planning to make a CNC machine using your design, and this improved accessibility to the design would greatly help me and I bet others would appreciate it too.

reefwarrior, 2019-10-13:
It would be greatly appreciated if this will be possible? :)

nunoadferreira, 2019-11-04:
Hi,

Thanks for sharing this great project. I loved it and I will definitely build one for myself. However, I have a prusa mini which can only print an area of 180 x 180 mm and the side panels are bigger than that. So I am thinking in splitting them to print in two pieces. It would be so useful if you could share the fusion 3d files or just the step files.

leonvandenbeukel, 2020-01-02:
The design is now available as a FreeCAD model (check the model folder).

closed 2020-01-02 by leonvandenbeukel