leonvandenbeukel/CNC-Machine-V3 #5: Fusion360 Model

Harriethijssen, 2020-01-08:
You mention you designed with Fusion360, but I only seem to find the FreeCad models. I'm not familiar with FreeCad (yet) and am wondering if there's another location where the Fusion360 models are.
I justed started playing with FreeCad, and it seems very similar to Fusion360. However, I need to make a few modifications to the designs and am not able to get to the "sketch" for the bodies in FreeCad. I'm sure that I'm overlooking some magic in FreeCad but did not find out yet.
Any chance you can upload the Fusion360 models (or share on the A360 cloud) or give me the magic sequence in FreeCad to make some updates to the design?

Thanks.

excessnet, 2020-01-31:
didnt try, but I guess you could just open in FreeCAD, export as .step file and then import into F360.

agehall, 2020-02-25:
It's pretty straight forward to export the Fusion 360 files - just go to File > Export and export the model as a F3D file. I too would appreciate that very much!

infosisio, 2021-07-13:
Just to clarify, the FreeCAD files are not the source files, they are only solids.

shablam, 2023-06-21:
Can you please provide the fusion 360 files or a parametric format? Great design but cant get into Solidworks without freecad. let this design fly bro!