leonvandenbeukel/CNC-Machine-V3 #6: Some Questions to Machine v3

rainerschulte, 2020-06-11:
Hi, I'm currently building the CNC V3 and found some Issues. 

1. The belts on Y-Axis cannot be adjusted, when fully assambled the X/Y-Axis because the Steppers are not raechable from below.
2. The Z-Axis Motor touches the upper rail when Using the 30 Teeth Pulleys.  
3. The most important Issue is selfmade. I built the mechanics on my Modeling Workbench with 1800 x 600 mm and the Steel-Rails are bending a bit. I'm affraid (or pretty shure) that it becomes worse when adding Dust collection. Any Idea how to fix this?

Petesystemspecialist, 2020-07-26:
Part of the Y mount STL file is missing!. Its the smaller slider

rainerschulte, 2020-07-28:
> 
> 
> Re CNC ver 3 . one of the 3 Y mount STL files is missing!. The Smaller slider is not shown. I'm stuck without it.

It's the "Universal Mount". It`s not labled seperately. These days my new Makita Router will arrive. I found my old little Dremel with the small Bits not being rigid enogh. I will test some new Collars from DREMEL for fine work. 
Funny... I like it, when it moves! Now I need to find out to find out how to change Tools in bCNC. Until here I'm happy with my CNC.

Petesystemspecialist, 2020-07-28:
Hi there

& thankyou for your help. Its a pity there are no images attached to the
parts.
Anyway, the 3d printer is hard at work again.
I like your comments re the Dremel. I was going to use this, but in view of
your comments,  I will get the larger ( Chinese copy :(
Best wishes

Peter Ford

Nudel, 2020-10-04:
>     3. The most important Issue is selfmade. I built the mechanics on my Modeling Workbench with 1800 x 600 mm and the Steel-Rails are bending a bit. I'm affraid (or pretty shure) that it becomes worse when adding Dust collection. Any Idea how to fix this?

I found some interesting discussions here: https://www.model-engineer.co.uk/forums/postings.asp?th=107717

My take is you could either get a thicker walled rail or hammer in a round rail to stiffen it up.

I would also consider filling it with concrete (or epoxy?), but I haven't had the time to find any definite answers if it would help that much.

The third option would require a redesign of the y axis motor mount. If you moved the motor to the opposite side of the x axis sidepanel you might be able to fit a support foot between the runners on the y axis rails. Just an idea.

> Hi there & thankyou for your help. Its a pity there are no images attached to the parts.
If you open the whole model in freecad you can select the individual parts to find the name for it, or select the component name and have it highlighted on the model. Maybe that could help a bit?

rainerschulte, 2020-10-04:
After some tries to mill something I decided to switch to a complete other Design. I had several Issues with Belt tensioning and found it very unpracticable to reach the Motors between Table and Mounts. My new CNC has much more build area and all Motors can be accessed without hassle. I have now 1000 x 430 mm available and I get usable outcome.
Sorry Leon.. But it was a good practice to learn.

closed 2020-10-04 by rainerschulte